rynbrd/xf86-input-mtrack #76: Two Finger Tap not working Macbook Air Trackpad

mrgalup, 2015-01-10:
Hello,

First off, this is an amazing driver that has completely fixed the lack of functionality for the Macbook Air Trackpad in Ubuntu 14.04 for me. There are only a few outstanding issues that I cannot solve. Two finger tap to right click is highly intermittent, to the degree where I sometimes have to forcefully tap on the trackpad, and it still only works about 60% of the time. Below are my customized settings:

	Option		"IgnoreThumb"		"1"
	Option		"IgnorePalm"		"1"
	Option		"TapDragEnable"		"0"
	Option		"FingerHigh"		"12"
	Option		"FingerLow"		"1"
	Option 		"Sensitivity"		"0.85"
	Option		"ButtonMoveEmulate"	"0"
	Option		"ClickFinger1"		"0"
	Option		"ClickFinger2"		"0"
	Option		"ClickFinger3"		"0"
	Option		"ClickTime"		"50"
#	Option		"ButtonTouchExpire"	"25"
	Option		"MaxTapMove"		"200"
	Option		"ScrollDistance"	"75"

The commented out "ButtonTouchExpire" option is to show that that didn't work either, however, I don't know if it is making any difference otherwise. Thanks for your help in advance and I hope this post helps others in my position!

Best,

BlueDragonX, 2015-01-10:
Many of the settings you have set there will have an adverse affect on the drivers ability to track multiple fibers for tapping. Remove FingerHigh, FingerLow,  and MaxTapMove and see if that improves things.

mrgalup, 2015-01-10:
Ryan,

Thank you so much for your speedy response and for your advice. Removed
these settings seems to have solved the problem!

Once again, superb work on this driver. I scoured the internet to find a
solution to the macbook air trackpad being practically useless, as well as
modifying bunches of synclient options to no avial. Kudos on the fine work
and I will recommend it to anyone I know.

Best,
Mark

closed 2015-01-10 by BlueDragonX

mrgalup, 2015-01-13:
Hello Ryan,

I apologize if this is an inconvenience, but I wondered if you had any
insight on settings which might help avoid my palm triggering tap clicks
while typing. I do some competitive coding and when I'm really moving I
find that my palm often will do this. My settings are below:

Option "IgnoreThumb" "1"
Option "IgnorePalm" "1"
Option "TapDragEnable" "0"
Option "Sensitivity" "0.85"
Option "ButtonMoveEmulate" "0"
Option "ClickFinger1" "0"
Option "ClickFinger2" "0"
Option "ClickFinger3" "0"
Option "ClickTime" "50"
Option "ScrollDistance" "50"

Thank you in advance for your time!

BlueDragonX, 2015-01-13:
Use dispad: https://github.com/BlueDragonX/dispad

It disables the trackpad while typing. Then you can set IgnoreThumb and IgnorePalm to 0.