rynbrd/xf86-input-mtrack #78: Delay between Swipes

JP-Ellis, 2015-01-11:
For most uses of swipes that I know of, only a single swipe is desired.  Currently, many swipes tend to activate in a row.  For example, I have swipes left and right which switch between workspaces; however, multiple swipes are frequently registered causing me to skip over workspaces.

Would it be possible to implement a delay beween swipes?  So that if a swipe is registered, a (configurable) 200ms have to pass before the next swipe can be triggered?

BlueDragonX, 2015-01-11:
There is no way to block further clicks. It works identically to scrolling as it's the same gesture with more fingers. You could try increasing SwipeDistance.

JP-Ellis, 2015-01-12:
SwipeDistance would have the desired effect if my swipes were consistently the same distance; however, I have tried various settings and what works one day will result in me skipping workspaces or me wondering why the workspace hasn't changed on the next day.

I am also curious as to how it is impossible to block further swipes (not any input...).  I would have thought something like
```
if time - last_swipe_time > cfg->swipeInterval { 
    last_swipe_time = now();
    return swipe;
}
````
should do the trick.

BlueDragonX, 2015-01-12:
Of course it's not impossible. It is just preferable to solve issues without a code change when possible.

That said you seem to have a solid grasp of how you'd like this implemented and I'm always willing to review and merge a pull request. I simply don't have the time to implement this at the moment.

newx, 2018-06-29:
@JP-Ellis can you share your config for workspace switch via swipe?