rynbrd/xf86-input-mtrack #79: Logitech T651 trackpad: clicking works, cursor movement does not

wbolster, 2015-01-31:
Using this `xorg.conf` snippet and `xserver-xorg-input-mtrack` version 0.3.0-1+b3 (Debian):

```
Section "InputClass"
    Identifier "trackpad"
    Driver "mtrack"

    # Natural scrolling direction
    Option "ButtonMapping" "1 2 3 5 4"
EndSection
```

... my Logitech T651 trackpad allows clicking (also double finger tap for "right click"), but the cursor won't move. (With the `evdev` driver cursor movement works.)

Relevant part of the Xorg logs:

```
[  9233.100] (II) config/udev: Adding input device Logitech T651 (/dev/input/event13)
[  9233.100] (**) Logitech T651: Applying InputClass "evdev pointer catchall"
[  9233.100] (**) Logitech T651: Applying InputClass "trackpad"
[  9233.100] (II) LoadModule: "mtrack"
[  9233.100] (II) Loading /usr/lib/xorg/modules/input/mtrack_drv.so
[  9233.101] (II) Module mtrack: vendor="X.Org Foundation"
[  9233.101]    compiled for 1.15.99.904, module version = 0.1.0
[  9233.101]    Module class: X.Org XInput Driver
[  9233.101]    ABI class: X.Org XInput driver, version 21.0
[  9233.101] (II) Using input driver 'mtrack' for 'Logitech T651'
[  9233.101] (**) Logitech T651: always reports core events
[  9233.101] (**) Logitech T651: always reports core events
[  9233.101] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:1a.1/usb2/2-2/2-2:1.0/bluetooth/hci0/hci0:11/0005:046D:B00C.0011/input/input30/event13"
[  9233.101] (II) XINPUT: Adding extended input device "Logitech T651" (type: TOUCHPAD, id 8)
[  9233.101] (II) device control: init
[  9233.101] (**) Option "Device" "/dev/input/event13"
[  9233.101] (II) mtrack: devname: Logitech T651
[  9233.101] (II) mtrack: devid: 46d b00c 500
[  9233.101] (II) mtrack: caps: left middle right
[  9233.124] (**) Logitech T651: (accel) keeping acceleration scheme 1
[  9233.124] (**) Logitech T651: (accel) acceleration profile 0
[  9233.124] (**) Logitech T651: (accel) acceleration factor: 2.000
[  9233.124] (**) Logitech T651: (accel) acceleration threshold: 4
[  9233.124] (II) device control: on
[  9233.124] (WW) Touchpad has minimal capabilities. Some features will be unavailable.
[  9233.125] (II) config/udev: Adding input device Logitech T651 (/dev/input/mouse1)
[  9233.125] (**) Logitech T651: Applying InputClass "trackpad"
[  9233.125] (II) Using input driver 'mtrack' for 'Logitech T651'
[  9233.125] (**) Logitech T651: always reports core events
[  9233.125] (**) Logitech T651: always reports core events
[  9233.125] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:1a.1/usb2/2-2/2-2:1.0/bluetooth/hci0/hci0:11/0005:046D:B00C.0011/input/input30/mouse1"
```

BlueDragonX, 2015-01-31:
It more like the kernel driver does not support multitouch capabilities for this device. Of this is the case it simply won't work.

You'll need to determine the kernel driver that trackpad is using and part that here.

wbolster, 2015-02-02:
Can you give me some pointers on how to get that information, @BlueDragonX? I'm rather new to this stuff... sorry. ;)

BlueDragonX, 2015-02-06:
Not entirely sure. See if the device shows up in `lsusb`. You may have to go spelunking in `/sys`, though.

wbolster, 2015-02-07:
`lsusb` output doesn't list it. (it's a bluetooth device connected through an usb bluetooth dongle.)

What I did find in `/sys/` is this:

```console
$ pwd
/sys/devices/pci0000:00/0000:00:1a.1/usb3/3-2/3-2:1.0/bluetooth/hci0/hci0:11/0005:046D:B00C.0003/input/input16

$ ls -1F
capabilities/
device@
event13/
id/
modalias
mouse1/
name
phys
power/
properties
subsystem@
uevent
uniq

$ cat name 
Logitech T651

$ cat uevent 
PRODUCT=5/46d/b00c/500
NAME="Logitech T651"
PHYS="00:1a:7d:0a:c5:a6"
UNIQ="00:1f:20:74:ea:3a"
PROP=0
EV=1f
KEY=70000 0 0 0 0
REL=143
ABS=10000000000
MSC=10
MODALIAS=input:b0005v046DpB00Ce0500-e0,1,2,3,4,k110,111,112,r0,1,6,8,a28,m4,lsfw

$ cat modalias 
input:b0005v046DpB00Ce0500-e0,1,2,3,4,k110,111,112,r0,1,6,8,a28,m4,lsfw

$ cd capabilities/

$ pwd
/sys/devices/pci0000:00/0000:00:1a.1/usb3/3-2/3-2:1.0/bluetooth/hci0/hci0:11/0005:046D:B00C.0003/input/input16/capabilities

$ ls -1F
abs
ev
ff
key
led
msc
rel
snd
sw

$ for f in *; do echo; echo $f; cat $f; done

abs
10000000000

ev
1f

ff
0

key
70000 0 0 0 0

led
0

msc
10

rel
143

snd
0

sw
0

$ cd ..

$ cd id/

$ pwd
 /sys/devices/pci0000:00/0000:00:1a.1/usb3/3-2/3-2:1.0/bluetooth/hci0/hci0:11/0005:046D:B00C.0003/input/input16/id

$ ls -1F
bustype
product
vendor
version

$ for f in *; do echo; echo $f; cat $f; done

bustype
0005

product
b00c

vendor
046d

version
0500
```

Is this of any use?