rynbrd/xf86-input-mtrack #80: Can't Build on Ubuntu 14.04.2

johnjelinek, 2015-02-21:
```
$ ./configure 
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking build system type... x86_64-unknown-linux-gnu
checking host system type... x86_64-unknown-linux-gnu
checking how to print strings... printf
checking for style of include used by make... GNU
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for a sed that does not truncate output... /bin/sed
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for fgrep... /bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking whether the shell understands some XSI constructs... yes
checking whether the shell understands "+="... yes
checking how to convert x86_64-unknown-linux-gnu file names to x86_64-unknown-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-unknown-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for mt... mt
checking if mt is a manifest tool... no
checking how to run the C preprocessor... gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for dlfcn.h... yes
checking for objdir... .libs
checking if gcc supports -fno-rtti -fno-exceptions... no
checking for gcc option to produce PIC... -fPIC -DPIC
checking if gcc PIC flag -fPIC -DPIC works... yes
checking if gcc static flag -static works... yes
checking if gcc supports -c -o file.o... yes
checking if gcc supports -c -o file.o... (cached) yes
checking whether the gcc linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking whether -lc should be explicitly linked in... no
checking dynamic linker characteristics... GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking whether stripping libraries is possible... yes
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... no
checking for gcc option to accept ISO C99... -std=gnu99
checking whether __clang__ is declared... no
checking whether __INTEL_COMPILER is declared... no
checking whether __SUNPRO_C is declared... no
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking if gcc -std=gnu99 supports -Werror=unknown-warning-option... no
checking if gcc -std=gnu99 supports -Werror=unused-command-line-argument... no
checking if gcc -std=gnu99 supports -Wall... yes
checking if gcc -std=gnu99 supports -Wpointer-arith... yes
checking if gcc -std=gnu99 supports -Wmissing-declarations... yes
checking if gcc -std=gnu99 supports -Wformat=2... yes
checking if gcc -std=gnu99 supports -Wstrict-prototypes... yes
checking if gcc -std=gnu99 supports -Wmissing-prototypes... yes
checking if gcc -std=gnu99 supports -Wnested-externs... yes
checking if gcc -std=gnu99 supports -Wbad-function-cast... yes
checking if gcc -std=gnu99 supports -Wold-style-definition... yes
checking if gcc -std=gnu99 supports -Wdeclaration-after-statement... yes
checking if gcc -std=gnu99 supports -Wunused... yes
checking if gcc -std=gnu99 supports -Wuninitialized... yes
checking if gcc -std=gnu99 supports -Wshadow... yes
checking if gcc -std=gnu99 supports -Wmissing-noreturn... yes
checking if gcc -std=gnu99 supports -Wmissing-format-attribute... yes
checking if gcc -std=gnu99 supports -Wredundant-decls... yes
checking if gcc -std=gnu99 supports -Wlogical-op... yes
checking if gcc -std=gnu99 supports -Werror=implicit... yes
checking if gcc -std=gnu99 supports -Werror=nonnull... yes
checking if gcc -std=gnu99 supports -Werror=init-self... yes
checking if gcc -std=gnu99 supports -Werror=main... yes
checking if gcc -std=gnu99 supports -Werror=missing-braces... yes
checking if gcc -std=gnu99 supports -Werror=sequence-point... yes
checking if gcc -std=gnu99 supports -Werror=return-type... yes
checking if gcc -std=gnu99 supports -Werror=trigraphs... yes
checking if gcc -std=gnu99 supports -Werror=array-bounds... yes
checking if gcc -std=gnu99 supports -Werror=write-strings... yes
checking if gcc -std=gnu99 supports -Werror=address... yes
checking if gcc -std=gnu99 supports -Werror=int-to-pointer-cast... yes
checking if gcc -std=gnu99 supports -Werror=pointer-to-int-cast... yes
checking if gcc -std=gnu99 supports -pedantic... yes
checking if gcc -std=gnu99 supports -Werror... yes
checking if gcc -std=gnu99 supports -Werror=attributes... yes
Package xorg-macros was not found in the pkg-config search path.
Perhaps you should add the directory containing `xorg-macros.pc'
to the PKG_CONFIG_PATH environment variable
No package 'xorg-macros' found
checking whether make supports nested variables... (cached) yes
checking for mtdev_open in -lmtdev... no
checking for atan2 in -lm... yes
checking for XORG... no
configure: error: Package requirements (xorg-server >= 1.7 xproto inputproto ) were not met:

No package 'xorg-server' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables XORG_CFLAGS
and XORG_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.
```

BlueDragonX, 2015-02-21:
You need to install xserver-xorg-dev. I determined this by a quick Google search which should probably be the first thing you do.

closed 2015-02-21 by BlueDragonX

johnjelinek, 2015-02-21:
What did you search for?

johnjelinek, 2015-02-21:
It didn't work.
```
$  sudo apt-get install xserver-xorg-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libcheese-gtk23 : Depends: libclutter-gtk-1.0-0 (>= 0.91.8) but it is not going to be installed
                   Depends: libcogl15 (>= 1.15.8) but it is not going to be installed
 libcheese7 : Depends: libclutter-gst-2.0-0 (>= 0.10.0) but it is not going to be installed
              Depends: gstreamer1.0-clutter but it is not going to be installed
 libclutter-1.0-0 : Depends: libcogl-pango15 (>= 1.15.8) but it is not going to be installed
                    Depends: libcogl15 (>= 1.15.8) but it is not going to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
```

johnjelinek, 2015-02-21:
`sudo apt-get install xserver-xorg-dev-lts-trusty` worked to get it to build