rynbrd/xf86-input-mtrack #81: Rightclick on Arch MacBook

DonHansDampf, 2015-02-21:
Hi there,

i installed your driver to use it on my MacBook Air 2013 with Archlinux. Till now it works good, i can use the Touchpad. But i cant change the Double-Tap-Gesture. I want, just like in MacOS, the Two-Finger-Tap (not click) as a right click. But it doenst matter how i configure the the driver, i always need a three-finger-tap to perfom a right click. 

Here is my xorg.conf under etc/X11/xorg.conf:

```
Section "InputClass"
    MatchIsTouchpad "on"
    Identifier      "Touchpads"
    MatchDevicePath "/dev/input/event10"
    Driver          "mtrack"
    Option      "TapButton2"    "2"
    Option      "TapButton3"    "3"
EndSection
```

I also tried TapButton2 = 3 and to configure it via /etc/X11/xorg.conf.d/10-mtrack.conf. Seems like i cant change the TapButton Settings. Do you have any tip for me?

tomauty, 2015-03-01:
+1

Experiencing the same issue. 

My current configuration:

```
Section "InputClass"
    MatchIsTouchpad "on"
    Identifier      "Touchpads"
    Driver          "mtrack"
    Option          "Sensitivity" "0.55"
    Option      "FingerHigh" "12"
    Option      "FingerLow" "1"
    Option          "IgnoreThumb" "true"
    Option          "IgnorePalm" "true"
    Option          "TapButton1" "0"
    Option          "TapButton2" "3"
    Option          "TapButton3" "3"
    Option          "ClickFinger1" "1"
    Option          "ClickFinger2" "3"
    Option          "ClickFinger3" "3"
    Option          "ButtonMoveEmulate" "false"
    Option      "ClickTime" "1"
    Option      "ButtonIntegrated" "true"
    Option          "ClickTime" "25"
    Option          "BottomEdge" "25"
    Option      "SwipeLeftButton" "8"
    Option      "SwipeRightButton" "9"
    Option      "SwipeUpButton" "0"
    Option      "SwipeDownButton" "0"
    Option      "ScrollDistance" "75"
    Option      "TapDragEnable" "false"
EndSection
```

tomauty, 2015-03-01:
I think the driver actually isn't loaded, or the config file isn't taking effect. I turned on `TrackpadDisable` and everything still works. What's the best way to check if this driver is active?

DonHansDampf, 2015-03-01:
When i remove the line "MatchDevicePath "/dev/input/event10"" my Touchpad doesnt work anymore. So it seems like it loads the config file... Mh.

tomauty, 2015-03-01:
Check out #69, adding those lines from the last comment, as well as re-installing `xf86-input-synaptics` and `xf86-input-mtrack` did the trick for me.

tomauty, 2015-03-01:
I'd also recommend just starting your configuration over from scratch if that works for you, to rule out other issues.

palmerj, 2015-04-11:
I'm also having exactly the same issue on my MacBookPro11,3 with 14.04.2. I also tried TapButton2 = 2 or 3 and it didn't make a difference after rebooting.

Here's my config:

```
Section "InputClass"
 MatchIsTouchpad "on"
 Identifier "Touchpads"
 Driver "mtrack"
 Option "IgnoreThumb" "true"
 Option "ThumbSize" "50"
 Option "IgnorePalm" "true"
 Option "DisableOnPalm" "false"
 Option "BottomEdge" "30"
 Option "TapDragEnable" "true"
 Option "Sensitivity" "0.4"
 Option "FingerHigh" "10"
 Option "FingerLow" "5"
 Option "ButtonEnable" "true"
 Option "ButtonIntegrated" "true"
 Option "ButtonMoveEmulate" "false"
 Option "ButtonTouchExpire" "750"
 Option "ClickFinger1" "1"
 Option "ClickFinger2" "3"
 Option "ClickFinger3" "2"
 Option "TapButton1" "1"
 Option "TapButton2" "3"
 Option "TapButton3" "2"
 Option "TapButton4" "0"
 Option "TapDragWait" "100"
 Option "ScrollLeftButton" "7"
 Option "ScrollRightButton" "6"
 Option "ScrollDistance" "100"
EndSection
```

garrett92895, 2015-08-09:
I've resolved the problem on my Macbook 11,1 with Arch and 4.1.4-1. For me, the code for a right click was 3. I noticed that, with `"ClickFinger2" "3"` I could successfully right-click with a three finger click. It dawned on me that perhaps the ClickFinger settings aren't actually considering the finger used to click. `"ClickFinger1"` seems to be the setting for clicking with another finger on the touchpad. Sure enough, setting `"ClickFinger1" "3"` fixed it for me. 

Here's my mtrack config:

    Section "InputClass"
     MatchIsTouchpad "on"
     Identifier "Touchpads"
     Driver "mtrack"
     Option "IgnoreThumb" "true"
     Option "ThumbSize" "50"
     Option "IgnorePalm" "true"
     Option "DisableOnPalm" "false"
     Option "BottomEdge" "30"
     Option "TapDragEnable" "true"
     Option "Sensitivity" "0.55"
     Option "FingerHigh" "3"
     Option "FingerLow" "2"
     Option "ButtonEnable" "true"
     Option "ButtonIntegrated" "true"
     Option "ButtonTouchExpire" "750"
     Option "ClickFinger1" "3"
     Option "ClickFinger2" "2"
     Option "TapButton1" "1"
     Option "TapButton2" "3"
     Option "TapButton3" "2"
     Option "TapDragWait" "100"
     Option "ScrollLeftButton" "7"
     Option "ScrollRightButton" "6"
     Option "ScrollDistance" "100"
    EndSection

To sum up the solutions in this thread...

**Ensure that the driver is being loaded properly**
`cat /var/log/Xorg.0.log | grep mtrack` should let you know whether or not it is being loaded properly. If you don't get any output, then it's not happening. Don't worry if you see `(EE) mtrack: cannot configure device` as it is most likely a red herring

**Ensure that your configuration is being loaded**
I did this by changing the Sensitivity and restarting my laptop (restarting Xserver probably would have been quicker). 

**Add user to the input group**
As brutasse pointed out in https://github.com/BlueDragonX/xf86-input-mtrack/issues/69#issuecomment-52619907(#69), because X is rootless it doesn't have permission to read `/dev/input/event10` by default. The solution is to add the user to the input group with `gpasswd -a <user> input`. If you don't want to do that, there is another solution available in that thread. 

**Set `"ClickFinger1"` to the appropriate value**
See my explanation above

closed 2015-08-10 by BlueDragonX