rynbrd/xf86-input-mtrack #83: mtrack driver does not appear to be loading

rcrath, 2015-03-20:
After a recent update, my mtrack driver does not appear when I run lsmod and I get the following result from 

```
cat Xorg.0.log | grep mtrack
[   100.765] (II) LoadModule: "mtrack"
[   100.766] (WW) Warning, couldn't open module mtrack
[   100.766] (II) UnloadModule: "mtrack"
[   100.766] (II) Unloading mtrack
[   100.766] (EE) Failed to load module "mtrack" (module does not exist, 0)
[   100.766] (EE) No input driver matching `mtrack'
[   185.049] (II) LoadModule: "mtrack"
[   185.050] (WW) Warning, couldn't open module mtrack
[   185.050] (II) UnloadModule: "mtrack"
[   185.050] (II) Unloading mtrack
[   185.050] (EE) Failed to load module "mtrack" (module does not exist, 0)
[   185.050] (EE) No input driver matching `mtrack'
log]

```

rcrath, 2015-03-22:
workaround was to uninstall mtrack for now.  hid_magicmouse then works fine and trackpad works.  The above problem occurred after kernel upgrade, so I am guessing it has something to do with mtrack not being built into the new kernel (not sure what the language is).  Still willing to debug if interested, just reopen issue.

closed 2015-03-22 by rcrath

BlueDragonX, 2015-03-22:
mtrack is not a kernel driver, it is an xorg driver. Your logs indicate that the driver could not be found which would indicate that it was installed incorrectly. You likely need to set --with-xorg-module-dir when calling ./configure.

rcrath, 2015-03-23:
ok, thanks!