rynbrd/xf86-input-mtrack #90: mtrack driver seems to not be working correctly on mid-2015 15 inch macbook pro

isomarcte, 2015-06-15:
Hey everyone,

I am running Linux on a mid 2015 15 inch macbook pro. I have been trying in vain to get the mtrack driver to work with the trackpad so that it behaves more like OS X. First off, I realize this may be a moot issue for the time being, because the trackpads on 2015 MBPs don't *officially* work *completely* with the current Linux kernel. Specifically, multitouch input is not detected right now. However I was able to fix the multitouch issue by recomiling the Linux kernel with the two patches proposed on this kernel bug issue.

That being said, while mutlitouch and two-finger scrolling is now working out of the box. I can't seem to configure it.

Here are my specific issues,
* Any changes to the my /etc/X11/xorg.conf.d/99-trackpad.conf file seem to have either **no effect at all* on the trackpad *or* they cause some sort of crash when I try to login to Gnome (after which I revert my changes in the file, restart the GDM and then I can log in again.) I have tried many changes in the conf file and most do nothing. One specific example of a change that will crash Gnome is setting the "TrackpadDisable" value to "3". I did this as an oracle to see if X was even reading my changes to this file. When I set this value to "3" Gnome will consistently crash on login.
* I use Gnome as my desktop environment, it detects a "touchpad" under the mouse settings, but it also detects a "mouse". Changes to the "touchpad" have no effect on anything as far as I can tell, however If I change the sensitivity on the "mouse" then the sensitivity on **both** a USB mouse and the actual trackpad are effected.

By the way, when I say that Gnome "crashed" what I really mean to say is that when I attempt to login to my profile, it accepts my passwords, starts login, then restarts the GDM as though I had never logged in. I am not actually sure what is happening there...but they best term I can use to describe it is a crash. System logs don't seem to show anything like the service restarting so I am somewhat at a loss.

Relevant Information

Kernel Bug: https://bugzilla.kernel.org/show_bug.cgi?id=96771
Patch 1: https://bugzilla.kernel.org/attachment.cgi?id=175481&action=diff
Patch 2: https://bugzilla.kernel.org/attachment.cgi?id=177881&action=diff

Linux: Linux foobar 4.0.5-1-mac #1 SMP PREEMPT Sat Jun 13 15:40:47 MDT 2015 x86_64 GNU/Linux
xorg-server: 1.17.1
Gnome:  GNOME Shell 3.16.2

If anyone has any ideas what is going on and how to fix it so I can configure the trackpad, I would be very appreciative. I am also just trying to get some help figuring out just where the issue is, i.e. Am I configuring mtrack incorrectly? Is Gnome screwing with the X configuration? Is the Linux patch not fully fixing the issues with the new trackpads (in which case I will report this upstream)? Or does mtrack just not yet fully support these new trackpads?

Also please forgive any grave mistakes I have made in this posting. I do not know much about X configuration.

Thanks in advance

isomarcte, 2015-06-15:
I believe I have found at least part of the issue.

I noticed this in the system log.

````
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (II) input device 'Apple Inc. Apple Internal Keyboard / Trackpad', /dev/input/event5 is a keyboard
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (II) config/udev: Adding input device bcm5974 (/dev/input/event11)
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (**) bcm5974: Applying InputClass "evdev touchpad catchall"
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (**) bcm5974: Applying InputClass "Touchpads"
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (**) bcm5974: Applying InputClass "touchpad catchall"
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (**) bcm5974: Applying InputClass "Default clickpad buttons"
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (**) bcm5974: Applying InputClass "Disable clickpad buttons on Apple touchpads"
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (**) bcm5974: Applying InputClass "libinput touchpad catchall"
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (**) bcm5974: Applying InputClass "Touchpads"
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (II) LoadModule: "mtrack"
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (II) Loading /usr/lib/xorg/modules/input/mtrack_drv.so
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (II) Module mtrack: vendor="X.Org Foundation"
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: compiled for 1.17.1, module version = 0.1.0
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: Module class: X.Org XInput Driver
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: ABI class: X.Org XInput driver, version 21.1
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (II) Using input driver 'mtrack' for 'bcm5974'
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (**) bcm5974: always reports core events
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (**) bcm5974: always reports core events
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (**) Option "FingerHigh" "5"
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (**) Option "FingerLow" "1"
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (**) Option "IgnoreThumb" "true"
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (**) Option "IgnorePalm" "true"
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (**) Option "DisableOnPalm" "true"
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (**) Option "BottomEdge" "30"
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (**) Option "ButtonIntegrated" "true"
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (**) Option "ClickFinger1" "1"
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (**) Option "ClickFinger2" "2"
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (**) Option "ClickFinger3" "3"
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (**) Option "ButtonMoveEmulate" "false"
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (**) Option "TapButton1" "1"
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (**) Option "TapButton2" "3"
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (**) Option "TapButton3" "2"
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (**) Option "TapButton4" "0"
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (**) Option "ClickTime" "25"
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (**) Option "ScrollDistance" "75"
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (**) Option "SwipeUpButton" "0"
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (**) Option "SwipeDownButton" "0"
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (**) Option "SwipeLeftButton" "8"
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (**) Option "SwipeRightButton" "9"
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (**) Option "Sensitivity" "0.64"
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-12/1-12:1.2/input/input11/event11"
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (II) XINPUT: Adding extended input device "bcm5974" (type: TOUCHPAD, id 12)
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (II) device control: init
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (**) Option "Device" "/dev/input/event11"
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (EE) xf86OpenSerial: Cannot open device /dev/input/event11
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: Permission denied.
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (EE) mtrack: cannot open device
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (EE) Couldn't init device "bcm5974"
Jun 14 18:15:35 foobar /usr/lib/gdm/gdm-x-session[9537]: (II) UnloadModule: "mtrack"
````

isomarcte, 2015-06-15:
Okay. Adding myself to the `input` group seems like it might have solved the issue. Any idea why I had to do that? Is that normal for `mtrack`?

BlueDragonX, 2015-06-15:
Your user needs access to the input devices. You have to be in the input group for that =)

garrett92895, 2015-08-09:
Because the newer versions of Xorg are rootless, they do not have permission to access input devices by default.

closed 2015-08-10 by BlueDragonX