rynbrd/xf86-input-mtrack #91: Support for 3-Finger Drag?

rb3, 2015-06-18:
Will there be support for 3-finger drag? I would really like to replicate the touchpad gestures on OS X. Thanks for this awesome project BTW! :)

p2rkw, 2015-07-15:
What you mean by 3 finger drag? How does is differ from one finger drag?

coxley, 2015-07-15:
Three fingers gesture in OS X begins click-hold, makes it convenient with no need to tap and such.

p2rkw, 2015-07-15:
Sound like something easy to add, especially on top of my recent changes, but what about swipes with 3 fingers? Does osx support them somehow? Is it able to recognize swipes and 3-finger dragging?

coxley, 2015-07-15:
Yeah, this shows some of the default swipe gestures OS X uses: https://support.apple.com/en-us/HT204895

This is three finger drag: https://support.apple.com/en-us/HT204609

Not sure if that answers your questions?

p2rkw, 2015-07-16:
Done. To test it build branch swipe-to-drag from here: https://github.com/p2rkw/xf86-input-mtrack/tree/swipe_to_drag

See bottom of readme for instructions how to enable it.

coxley, 2015-07-17:
Nice! Seems to work as you'd expect, at least for the 5 minute impression. :)

p2rkw, 2015-07-18:
Let me give you a tip: using swipe3 for button "1" drag is such a waste ;)
Firstly: it duplicates tap-to-drag functionality (unless someone will make tap-to-drag button configurable :) )
Secondly: mtrack can be easily configured to work with [easystroke](https://github.com/thjaeger/easystroke/wiki):
  configure swipe-2-drag as usual but instead of button 1 use eg. button 8:
```
        Option "SwipeDistance" "1"
        Option "SwipeLeftButton" "8"
        Option "SwipeRightButton" "8"
        Option "SwipeUpButton" "8"
        Option "SwipeDownButton" "8"
        Option "SwipeClickTime" "0"
        Option "SwipeSensitivity" "1100"
```
or same thing using `xinput`:
```
xinput set-prop 11 "Trackpad Swipe Buttons"  8 8 8 8
xinput set-prop 11 "Trackpad Swipe Settings" 1, 0, 1100
```
Just replace "11" with your touchpad id.

Next run easystroke, go to settings tab, and set gesture button to 8 (without modifiers).
Now every time you swipe with 3 fingers easystroke will recognize it as gesture :D

You can additionally configure Swipe4 or Scroll swipes to send button 1 events, so for example 
you can use Swipe in conjunction with easystroke and Scroll or Swipe4 (two or four finger swipes) for text selection.

[1] https://github.com/thjaeger/easystroke/wiki

akatrevorjay, 2016-05-08:
Hi! Was this merged into master by chance?