rynbrd/xf86-input-mtrack #102: Two finger scroll continues to scroll after movement stops

japostoles, 2015-11-29:
I'm having an issue where, when using two finger scroll, I'll stop moving and the application keeps scrolling.  This happens across apps and is not application specific.

Steps to reproduce (using latest master)

1. Start two-finger scrolling
2. Stop moving hand but keep fingers on trackpad

Page continues to scroll until fingers are lifted.

Macbook Air 2011 / Arch Linux

BlueDragonX, 2015-11-30:
I'll need a few things in order to debug this:
- Your trackpad configuration in xorg.conf.
- Debug output of this happening. You'll need to recompile the driver with all debuging enabled.

cccplex, 2015-12-13:
I have the same issue. It started after I changed the default config file with this one:

        Section "InputClass"
            MatchIsTouchpad "on"
            Identifier      "Touchpads"
            Driver          "mtrack"
            Option          "Sensitivity" "0.65"
            Option          "IgnoreThumb" "true"
            Option          "IgnorePalm" "true"
            Option          "TapButton1" "1"
            Option          "TapButton2" "3"
            Option          "TapButton3" "2"
            Option          "ClickFinger1" "1"
            Option          "ClickFinger2" "3"
            Option          "ClickFinger3" "2"
            Option          "BottomEdge" "25"
        EndSection

I haven't recompiled with debugging enabled. But maybe that error in the end is related? 

    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (II) config/udev: Adding input device bcm5974 (/dev/input/event15)
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (**) bcm5974: Applying InputClass "evdev touchpad catchall"
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (**) bcm5974: Applying InputClass "Touchpads"
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (II) LoadModule: "mtrack"
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (II) Loading /usr/lib/xorg/modules/input/mtrack_drv.so
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (II) Module mtrack: vendor="X.Org Foundation"
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: compiled for 1.18.0, module version = 0.2.0
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: Module class: X.Org XInput Driver
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: ABI class: X.Org XInput driver, version 22.1
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (II) Using input driver 'mtrack' for 'bcm5974'
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (**) bcm5974: always reports core events
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (**) bcm5974: always reports core events
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (**) Option "IgnoreThumb" "true"
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (**) Option "IgnorePalm" "true"
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (**) Option "ClickFinger1" "1"
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (**) Option "ClickFinger2" "3"
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (**) Option "ClickFinger3" "2"
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (**) Option "TapButton1" "1"
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (**) Option "TapButton2" "3"
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (**) Option "TapButton3" "2"
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (**) Option "Sensitivity" "0.65"
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:1a.7/
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (II) XINPUT: Adding extended input device "bcm5974" (type: TOUCHPAD, 
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (II) device control: init
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (**) Option "Device" "/dev/input/event15"
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (II) mtrack: devname: bcm5974
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (II) mtrack: devid: 5ac 253 1
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (II) mtrack: caps: left mtdata ibt
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (II) mtrack: 0: min: 0 max: 2048
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (II) mtrack: 1: min: 0 max: 2048
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (II) mtrack: 2: min: 0 max: 2048
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (II) mtrack: 3: min: 0 max: 2048
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (II) mtrack: 4: min: -16384 max: 16384
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (II) mtrack: 5: min: -4750 max: 5280
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (II) mtrack: 6: min: -150 max: 6730
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (II) mtrack: 9: min: 0 max: 65535
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (**) bcm5974: (accel) keeping acceleration scheme 1
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (**) bcm5974: (accel) acceleration profile 0
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (**) bcm5974: (accel) acceleration factor: 2.000
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (**) bcm5974: (accel) acceleration threshold: 4
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (II) device control: on
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (II) Touchpad supports regular and approaching touches.
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (II)   touch_min = 0, touch_max = 2048
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (II) Touchpad supports minor touch widths.
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (II) config/udev: Adding input device bcm5974 (/dev/input/mouse1)
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (**) bcm5974: Applying InputClass "Touchpads"
[... cut ...]
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (II) XINPUT: Adding extended input device "bcm5974" (type: TOUCHPAD, 
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (II) device control: init
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (**) Option "Device" "/dev/input/mouse1"
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (EE) mtrack: cannot configure device
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (EE) Couldn't init device "bcm5974"
    Dec 05 08:52:16 flappie /usr/lib/gdm/gdm-x-session[440]: (II) UnloadModule: "mtrack"

mro95, 2016-01-23:
Same issue here. 
Where can I enable debuging mode?

Here is my 00-touchpad.conf
```
Section "InputClass"
    MatchIsTouchpad "on"
    Identifier      "Touchpads"
    Driver          "mtrack"
    Option          "Sensitivity" "0.95"
    Option          "FingerHigh" "12"
    Option          "FingerLow" "1"
    Option          "IgnoreThumb" "true"
    Option          "IgnorePalm" "true"
    Option          "DisableOnPalm" "true"
    Option          "TapButton1" "1"
    Option          "TapButton2" "3"
    Option          "TapButton3" "2"
    Option          "TapButton4" "0"
    Option          "ClickFinger1" "1"
    Option          "ClickFinger2" "3"
    Option          "ClickFinger3" "3"
    Option          "ButtonMoveEmulate" "false"
    Option          "ButtonIntegrated" "true"
    Option          "ClickTime" "25"
    Option          "BottomEdge" "25"
    Option          "SwipeLeftButton" "8"
    Option          "SwipeRightButton" "9"
    Option          "SwipeUpButton" "0"
    Option          "SwipeDownButton" "0"
    Option          "ScrollDistance" "75"
    Option          "VertScrollDelta" "-111"
    Option          "HorizScrollDelta" "-111"
EndSection
```

mro95, 2016-02-03:
Still not working...

Debug Output
```
  touching p(-251, 1337) d(%+d, %+d) dir(4.78) down(1453560875541)
(II) trigger_move: -142, 201
(II) is_thumb: no  83 > 70 && 6 > 25
(II) is_palm: no  4 > 40
(II) mtstate: 1 touches at event time 1453560875645 (rt 1453560875645)
(II) (EE) BUG: triggered 'if (f[f_idx])'
(EE) BUG: log.c:495 in vpnprintf()
(EE) Unsupported printf directive '+'
(EE) 
(EE) Backtrace:
(EE) 0: /usr/lib/xorg-server/Xorg (ErrorFSigSafe+0xa15) [0x5a12f5]
(EE) 1: /usr/lib/xorg-server/Xorg (LogVMessageVerbSigSafe+0x81) [0x5a1ae1]
(EE) 2: /usr/lib/xorg-server/Xorg (LogVMessageVerb+0xb5) [0x5a1d45]
(EE) 3: /usr/lib/xorg-server/Xorg (xf86Msg+0x9c) [0x4823dc]
(EE) 4: /usr/lib/xorg/modules/input/mtrack_drv.so (_init+0x6715) [0x7fb8767ed7b5]
(EE) 5: /usr/lib/xorg/modules/input/mtrack_drv.so (_init+0x5c6b) [0x7fb8767ecbcb]
(EE) 6: /usr/lib/xorg/modules/input/mtrack_drv.so (_init+0x73ef) [0x7fb8767efadf]
(EE) 7: /usr/lib/xorg-server/Xorg (DPMSSupported+0xe8) [0x476248]
(EE) 8: /usr/lib/xorg-server/Xorg (xf86SerialModemClearBits+0x2b2) [0x49f982]
(EE) 9: /usr/lib/libc.so.6 (__restore_rt+0x0) [0x7fb87f24767f]
(EE) 10: /usr/lib/libc.so.6 (__select_nocancel+0x2a) [0x7fb87f2f5e1a]
(EE) 11: /usr/lib/xorg-server/Xorg (WaitForSomething+0x1d7) [0x590c97]
(EE) 12: /usr/lib/xorg-server/Xorg (SendErrorToClient+0x10e) [0x43609e]
(EE) 13: /usr/lib/xorg-server/Xorg (remove_fs_handlers+0x453) [0x43a293]
(EE) 14: /usr/lib/libc.so.6 (__libc_start_main+0xf0) [0x7fb87f234610]
(EE) 15: /usr/lib/xorg-server/Xorg (_start+0x29) [0x4245c9]
(EE) 16: ? (?+0x29) [0x29]
(EE) 
(EE) BUG: triggered 'if (f[f_idx])'
(EE) BUG: log.c:495 in vpnprintf()
(EE) Unsupported printf directive '+'
(EE) 
(EE) Backtrace:
(EE) 0: /usr/lib/xorg-server/Xorg (ErrorFSigSafe+0xa15) [0x5a12f5]
(EE) 1: /usr/lib/xorg-server/Xorg (LogVMessageVerbSigSafe+0x81) [0x5a1ae1]
(EE) 2: /usr/lib/xorg-server/Xorg (LogVMessageVerb+0xb5) [0x5a1d45]
(EE) 3: /usr/lib/xorg-server/Xorg (xf86Msg+0x9c) [0x4823dc]
(EE) 4: /usr/lib/xorg/modules/input/mtrack_drv.so (_init+0x6715) [0x7fb8767ed7b5]
(EE) 5: /usr/lib/xorg/modules/input/mtrack_drv.so (_init+0x5c6b) [0x7fb8767ecbcb]
(EE) 6: /usr/lib/xorg/modules/input/mtrack_drv.so (_init+0x73ef) [0x7fb8767efadf]
(EE) 7: /usr/lib/xorg-server/Xorg (DPMSSupported+0xe8) [0x476248]
(EE) 8: /usr/lib/xorg-server/Xorg (xf86SerialModemClearBits+0x2b2) [0x49f982]
(EE) 9: /usr/lib/libc.so.6 (__restore_rt+0x0) [0x7fb87f24767f]
(EE) 10: /usr/lib/libc.so.6 (__select_nocancel+0x2a) [0x7fb87f2f5e1a]
(EE) 11: /usr/lib/xorg-server/Xorg (WaitForSomething+0x1d7) [0x590c97]
(EE) 12: /usr/lib/xorg-server/Xorg (SendErrorToClient+0x10e) [0x43609e]
(EE) 13: /usr/lib/xorg-server/Xorg (remove_fs_handlers+0x453) [0x43a293]
(EE) 14: /usr/lib/libc.so.6 (__libc_start_main+0xf0) [0x7fb87f234610]
(EE) 15: /usr/lib/xorg-server/Xorg (_start+0x29) [0x4245c9]
(EE) 16: ? (?+0x29) [0x29]
(EE) 
```