rynbrd/xf86-input-mtrack #103: Bottom-right corner right-click

alfredxing, 2016-01-10:
Hi!
I'm trying to set up my MacBook Air (early 2014) on Arch, but I can't seem to find a way to get the driver to emulate a right-click when I click on the bottom right corner of the trackpad. Is the button zones feature the correct way to achieve this?

My current config looks like this:
```
Section "InputClass"
  Identifier "touchpad"
  Driver "mtrack"
  MatchIsTouchpad "on"
  MatchDevicePath "/dev/input/event*"

  Option "Sensitivity" "0.65"

  Option "TapButton1" "1"
  Option "TapButton2" "0"
  Option "TapButton3" "0"

  Option "ClickFinger1" "1"
  Option "ClickFinger2" "2"

  Option "ButtonIntegrated" "true"
  Option "ButtonZonesEnabled" "true"

  Option "IgnoreThumb" "true"
  Option "IgnorePalm" "true"
EndSection
```

BlueDragonX, 2016-01-10:
Yes. A right click is button 3 not 2 as you have it configured.

alfredxing, 2016-01-10:
Oh, okay, thanks!
Though after changing `ClickFinger2` to `"3"` it still doesn't seem to be working.

zgrep, 2016-03-09:
As far as I can tell (from the documentation) `ButtonZonesEnable` enabled it only to the effect that it will recognize `ClickFinger2` in the second zone and `ClickFinger1` in the first zone. This means that it will recognize a two-finger click in the second zone as what you want. At least, this is what happens for me.