rynbrd/xf86-input-mtrack #104: scrolling speed is non-configurable

jimdigriz, 2016-01-12:
Just moved from the multitouch to mtrack driver to solve some problems for my Microsoft Surface Pro 4, and found that though pretty much everything is better than the old driver, I find the two-finger scrolling is about ten times slower and unbearably slow, at least to my liking.

I dug around trying to find out how to increase the sensitivity, either via X (xinput does not have a suitable property) or via the mtrack driver its-self (ie. InputClass options), and came up with nothing.

I can adjust the mouse sensivity (Device Accel...), but it is the scrolling speed I wish to change and this seems not possible with the mtrack driver (or the multitouch one, though the speed there was fine).

Am I doing something stupid, or is this functionality actually missing?

```
alex@quatermain:~$ xinput 
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ Microsoft Surface Type Cover Mouse        id=11   [slave  pointer  (2)]
⎜   ↳ Microsoft Surface Type Cover Consumer Control     id=12   [slave  pointer  (2)]
⎜   ↳ Microsoft Surface Type Cover UNKNOWN      id=13   [slave  pointer  (2)]
⎜   ↳ Surface Pen                               id=14   [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳ Power Button                              id=6    [slave  keyboard (3)]
    ↳ Video Bus                                 id=7    [slave  keyboard (3)]
    ↳ Surface Pro Series Buttons                id=8    [slave  keyboard (3)]
    ↳ Sleep Button                              id=9    [slave  keyboard (3)]
    ↳ Microsoft Surface Type Cover Keyboard     id=10   [slave  keyboard (3)]

alex@quatermain:~$ xinput list-props 13
Device 'Microsoft Surface Type Cover UNKNOWN':
        Device Enabled (137):   1
        Coordinate Transformation Matrix (139): 1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
        Device Accel Profile (266):     0
        Device Accel Constant Deceleration (267):       1.000000
        Device Accel Adaptive Deceleration (268):       1.000000
        Device Accel Velocity Scaling (269):    10.000000
        Trackpad Disable Input (296):   0
        Trackpad Sensitivity (297):     0.250000
        Trackpad Touch Pressure (298):  7, 7
        Trackpad Button Settings (299): 1, 1
        Trackpad Button Emulation Settings (300):       0, 1, 100
        Trackpad Button Emulation Values (301): 3, 2, 0
        Trackpad Tap Settings (302):    70, 120, 400
        Trackpad Tap Button Emulation (303):    1, 3, 0, 0
        Trackpad Thumb Detection (304): 1, 0
        Trackpad Thumb Size (305):      25, 70
        Trackpad Palm Detection (306):  1, 0
        Trackpad Palm Size (307):       40
        Trackpad Gesture Settings (308):        10, 100
        Trackpad Scroll Distance (309): 150
        Trackpad Scroll Buttons (310):  5, 4, 7, 6
        Trackpad Swipe Distance (311):  1400
        Trackpad Swipe Buttons (312):   0, 0, 9, 8
        Trackpad Swipe4 Distance (313): 700
        Trackpad Swipe4 Buttons (314):  0, 0, 0, 0
        Trackpad Scale Distance (315):  150
        Trackpad Scale Buttons (316):   14, 15
        Trackpad Rotate Distance (317): 150
        Trackpad Drag Settings (318):   1, 350, 30, 200
        Trackpad Axis Inversion (319):  0, 0
```

louispvb, 2016-02-19:
I am also having this issue

louispvb, 2016-02-19:
xinput set-prop 13 309 25

BlueDragonX, 2016-02-21:
Use scroll-distance. A smaller value is faster.

closed 2016-02-21 by BlueDragonX