rynbrd/xf86-input-mtrack #105: Apple's Magic Trackpad 2

atotic, 2016-01-26:
I've just been forced to use a beastly Ubuntu 14.04 desktop at work. I missed my trackpad so much, I've bought Apple's Magic Trackpad 2, and I've been trying to make it work many hours. From what I understand, mtrack is the way to do it, I just can't make xorg use mtrack as the driver.

I've compiled mtrack from scratch, and installed it. It is in the Xorg's drivers directory.
Unfortunately, Xorg insists on pairing it to evdev:
```
cat /var/log/Xorg.0.log | grep Trackpad
[ 46057.101] config/udev: Adding input device Apple Trackpad (/dev/input/event14)
[ 46057.101] (**) Apple Trackpad: Applying InputClass "evdev pointer catchall"
[ 46057.101] (II) Using input driver 'evdev' for 'Apple Trackpad'
[ 46057.101] (**) Apple Trackpad: always reports core events
```
I've been toying with mtrack-conf. Can anyone point me in the right direction. If anyone is a consultant, I'd be happy to pay for a 30 min session, I know how annoying these newbie help requests can be. And you'll save me from the carpal tunnel...

```
xorg.conf.d> cat 09-mtrack.conf 
Section "Module"
	Load "mtrack"
EndSection

Section "InputClass"
	MatchIsTouchpad "true"
	Identifier "Multitouch Touchpad Trackpad"
	Driver "mtrack"
	Option          "Sensitivity" "0.95"
        Option          "FingerHigh" "12"
        Option          "FingerLow" "1"
        Option          "IgnoreThumb" "true"
        Option          "IgnorePalm" "true"
        Option          "TapButton1" "0"
        Option          "TapButton2" "0"
        Option          "TapButton3" "0"
        Option          "TapButton4" "0"
        Option          "ClickFinger1" "1"
        Option          "ClickFinger2" "3"
        Option          "ClickFinger3" "3"
        Option          "ButtonMoveEmulate" "false"
        Option          "ButtonIntegrated" "true"
        Option          "ClickTime" "25"
        Option          "BottomEdge" "25"
        Option          "SwipeLeftButton" "8"
        Option          "SwipeRightButton" "9"
        Option          "SwipeUpButton" "0"
        Option          "SwipeDownButton" "0"
        Option          "ScrollDistance" "75"
        Option          "ScrollUpButton" "5"
        Option          "ScrollDownButton" "4"
EndSection
```

atotic, 2016-02-06:
the problem is that udev is not detecting the trackpad properly, and mtrack never gets loaded.

closed 2016-02-06 by atotic

sbelov, 2017-07-02:
In case you still haven't solved this, see
- https://github.com/tiziano88/magic-trackpad-2-linux/issues/4
- https://github.com/robbi5/magictrackpad2-dkms
This driver worked for me.