rynbrd/xf86-input-mtrack #106: All works, but swiping

LordTaifleh, 2016-02-14:
Hello there, I hope you have a nice day!

I got mtrack to work with my touchpad and all changes in the .conf file are reflected after a login.
Xev properly delivers feedback on the emulated button presses.
As of now, clicking, 1,2,3 finger tapping, palm detect, 2 finger scroll, rotate and pinch work more or less and deliver appropriate button presses in xev. 
However, no matter how many fingers I use to swipe (well, more than 1, obviously), xev only recognizes the usual two finger scroll button presses.
I think something might be misconfigured here, but i cant figure out what it is, as it also behaves that way on default settings.
Any help would be appreciated.

`Section "InputClass"`
`    MatchIsTouchpad "on"`
 `   MatchProduct "SynPS/2 Synaptics TouchPad"`
  `  MatchDevicePath "/dev/input/event*"`
   ` Identifier "mtrack touchpads"`
   ` Driver "mtrack"`
`Option "FingerHigh" "5"`
`Option "FingerLow" "5"`
`Option "IgnoreThumb" "on" `
`Option "IgnorePalm" "on" `
`Option "DisableOnThumb" "off"`
`Option "DisableOnPalm" "off"`
`Option "ThumbRatio" "70"`
`Option "ThumbSize" "25"`
`Option "PalmSize" "40"`
`Option "EdgeSize" "10"`
`Option "ButtonEnable" "on"`
`Option "ButtonIntegrated" "on"`
`Option "ButtonTouchExpire" "100"`
`Option "ButtonZonesEnable" "true"`
`Option "ClickFinger1" "1"`
`Option "ClickFinger2" "3"`
`Option "ClickFinger3" "0"`
`Option "ButtonMoveEmulate" "on"`
`Option "TapButton1" "1"`
`Option "TapButton2" "3"`
`Option "TapButton3" "2"`
`Option "TapButton4" "0"`
`Option "ClickTime" "50"`
`Option "MaxTapTime" "120"`
`Option "MaxTapMove" "300"     # defaults 450`
`Option "GestureClickTime" "10"`
`Option "GestureWaitTime" "100"`
`Option "ScrollSmooth" "0"`
`Option "ScrollDistance" "150"`
`Option "ScrollClickTime" "20"`
`Option "ScrollSensitivity" "0"`
`Option "ScrollUpButton" "4"`
`Option "ScrollDownButton" "5"`
`Option "ScrollLeftButton" "6"`
`Option "ScrollRightButton" "7"`
`Option "ScrollCoastEnableSpeed" "0.5"`
`Option "ScrollCoastDuration" "200.0"`
`Option "SwipeDistance" "80"`
`Option "SwipeClickTime" "300"`
`Option "SwipeSensitivity" "1"`
`Option "SwipeUpButton" "0"`
`Option "SwipeDownButton" "0"`
`Option "SwipeLeftButton" "8"`
`Option "SwipeRightButton" "9"`
`Option "Swipe4Distance" "80"`
`Option "Swipe4ClickTime" "300"`
`Option "Swipe4Sensitivity" "1"`
`Option "Swipe4UpButton" "0"`
`Option "Swipe4DownButton" "0"`
`Option "Swipe4LeftButton" "8"`
`Option "Swipe4RightButton" "9"`
`Option "ScaleDistance" "80"`
`Option "ScaleUpButton" "12"`
`Option "ScaleDownButton" "13"`
`Option "RotateDistance" "0"        #defaults "150"`
`Option "RotateLeftButton" "0"       #defaults "14"`
`Option "RotateRightButton" "0"        #defaults "15"`
`Option "Hold1Move1StationaryButton" "1"`
`Option "Hold1Move1StationaryMaxMove" "1"`
`Option "TapDragEnable" "on"`
`Option "TapDragTime" "350"`
`Option "TapDragWait" "30"`
`Option "TapDragDist" "200"`
`Option "AxisXInvert" "false"`
`Option "AxisYInvert" "false"`
`Option "Sensitivity" "1"`
`EndSection`

BlueDragonX, 2016-02-14:
Please include your Xorg.0.log.

LordTaifleh, 2016-02-15:
this is Xorg.0.log at ~/.local/share/xorg
[    17.466] 
X.Org X Server 1.18.1
Release Date: 2016-02-08
[    17.466] X Protocol Version 11, Revision 0
[    17.466] Build Operating System: Linux 4.2.5-1-ARCH x86_64 
[    17.467] Current Operating System: Linux alphakevin 4.4.1-2-ARCH #1 SMP PREEMPT Wed Feb 3 13:12:33 UTC 2016 x86_64
[    17.467] Kernel command line: BOOT_IMAGE=/vmlinuz-linux root=UUID=aecb42b6-d8ea-441d-96ba-9533d050eb16 rw intel_pstate=disable kdbus=1
[    17.467] Build Date: 11 February 2016  09:57:03AM
[    17.467]  
[    17.467] Current version of pixman: 0.34.0
[    17.467] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[    17.467] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    17.468] (==) Log file: "/home/omicron/.local/share/xorg/Xorg.0.log", Time: Mon Feb 15 12:34:51 2016
[    17.478] (==) Using config directory: "/etc/X11/xorg.conf.d"
[    17.479] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    17.484] (==) No Layout section.  Using the first Screen section.
[    17.484] (**) |-->Screen "Screen0" (0)
[    17.484] (**) |   |-->Monitor "Monitor0"
[    17.484] (==) No device specified for screen "Screen0".
	Using the first device section listed.
[    17.484] (**) |   |-->Device "Screen0"
[    17.484] (==) Automatically adding devices
[    17.484] (==) Automatically enabling devices
[    17.484] (==) Automatically adding GPU devices
[    17.484] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    17.486] (WW) The directory "/usr/share/fonts/Type1/" does not exist.
[    17.486] 	Entry deleted from font path.
[    17.486] (WW) `fonts.dir' not found (or not valid) in "/usr/share/fonts/75dpi/".
[    17.486] 	Entry deleted from font path.
[    17.486] 	(Run 'mkfontdir' on "/usr/share/fonts/75dpi/").
[    17.486] (==) FontPath set to:
	/usr/share/fonts/misc/,
	/usr/share/fonts/TTF/,
	/usr/share/fonts/OTF/,
	/usr/share/fonts/100dpi/
[    17.486] (==) ModulePath set to "/usr/lib/xorg/modules"
[    17.486] (II) The server relies on udev to provide the list of input devices.
	If no devices become available, reconfigure udev or disable AutoAddDevices.
[    17.487] (II) Loader magic: 0x81ad60
[    17.487] (II) Module ABI versions:
[    17.487] 	X.Org ANSI C Emulation: 0.4
[    17.487] 	X.Org Video Driver: 20.0
[    17.487] 	X.Org XInput driver : 22.1
[    17.487] 	X.Org Server Extension : 9.0
[    17.488] (++) using VT number 1

[    17.488] (--) controlling tty is VT number 1, auto-enabling KeepTty
[    17.492] (II) systemd-logind: took control of session /org/freedesktop/login1/session/c1
[    17.493] (II) xfree86: Adding drm device (/dev/dri/card0)
[    17.494] (II) systemd-logind: got fd for /dev/dri/card0 226:0 fd 8 paused 0
[    17.497] (--) PCI:*(0:0:2:0) 8086:0416:17aa:3801 rev 6, Mem @ 0xc2000000/4194304, 0xb0000000/268435456, I/O @ 0x00005000/64
[    17.497] (--) PCI: (0:1:0:0) 10de:0fcd:17aa:3801 rev 161, Mem @ 0xc0000000/16777216, 0x90000000/268435456, 0xa0000000/33554432, I/O @ 0x00004000/128, BIOS @ 0x????????/524288
[    17.497] (II) Open ACPI successful (/var/run/acpid.socket)
[    17.497] (II) LoadModule: "glx"
[    17.499] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[    17.514] (II) Module glx: vendor="X.Org Foundation"
[    17.514] 	compiled for 1.18.1, module version = 1.0.0
[    17.514] 	ABI class: X.Org Server Extension, version 9.0
[    17.514] (==) AIGLX enabled
[    17.514] (II) LoadModule: "intel"
[    17.515] (II) Loading /usr/lib/xorg/modules/drivers/intel_drv.so
[    17.520] (II) Module intel: vendor="X.Org Foundation"
[    17.520] 	compiled for 1.18.0, module version = 2.99.917
[    17.520] 	Module class: X.Org Video Driver
[    17.520] 	ABI class: X.Org Video Driver, version 20.0
[    17.520] (II) intel: Driver for Intel(R) Integrated Graphics Chipsets:
[... 380 lines cut ...]
[  1530.930] (**) Option "_source" "server/udev"
[  1530.956] (II) failed to create input device '/dev/input/mouse0'.
[  1530.956] (EE) libinput: Logitech M705: Failed to create a device for /dev/input/mouse0
[  1530.956] (EE) PreInit returned 2 for "Logitech M705"
[  1530.956] (II) UnloadModule: "libinput"
[  1531.011] (II) config/udev: Adding input device Logitech M705 (/dev/input/event11)
[  1531.011] (**) Logitech M705: Applying InputClass "libinput pointer catchall"
[  1531.011] (**) Logitech M705: Applying InputClass "M705 libinput"
[  1531.013] (II) systemd-logind: got fd for /dev/input/event11 13:75 fd 54 paused 0
[  1531.013] (II) Using input driver 'libinput' for 'Logitech M705'
[  1531.013] (**) Logitech M705: always reports core events
[  1531.013] (**) Option "Device" "/dev/input/event11"
[  1531.013] (**) Option "_source" "server/udev"
[  1531.014] (II) input device 'Logitech M705', /dev/input/event11 is tagged by udev as: Mouse
[  1531.014] (II) Device 'Logitech M705' set to 800 DPI
[  1531.014] (II) input device 'Logitech M705', /dev/input/event11 is a pointer caps
[  1531.014] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.2/0003:046D:C52B.0007/0003:046D:101B.0008/input/input19/event11"
[  1531.014] (II) XINPUT: Adding extended input device "Logitech M705" (type: MOUSE, id 9)
[  1531.014] (**) Option "AccelerationScheme" "none"
[  1531.015] (**) Logitech M705: (accel) selected scheme none/0
[  1531.015] (**) Logitech M705: (accel) acceleration factor: 2.000
[  1531.015] (**) Logitech M705: (accel) acceleration threshold: 4
[  1531.016] (II) input device 'Logitech M705', /dev/input/event11 is tagged by udev as: Mouse
[  1531.016] (II) Device 'Logitech M705' set to 800 DPI
[  1531.016] (II) input device 'Logitech M705', /dev/input/event11 is a pointer caps

BlueDragonX, 2016-02-21:
You're using an old school Synaptics pressure sensitive trackpad. They're not true multitouch. They only support tracking two fingers at a time. Three or four finger swipes won't work.

closed 2016-02-21 by BlueDragonX

LordTaifleh, 2016-03-02:
well it works in windows, and is built into a 2014 model lenovo y510p so i don't think its a trackpad as you described. why would three finger tapping work then?