rynbrd/xf86-input-mtrack #107: Erroneous clicking

blcs79, 2016-03-01:
On my MacBook Pro (mid-2012 non-retina), I have tap-to-click disabled and want to only use the hardware button for clicking. However, sometimes when my thumb plus another finger is touching the trackpad, an erroneous click occurs. This makes using the trackpad very unreliable. Any idea why this happens? Using Fedora 23 btw.
Here's my settings:

Section "InputClass"
 MatchIsTouchpad "on"
 Identifier "Touchpads"
 Driver "mtrack"
 Option "IgnoreThumb" "true"
 Option "IgnorePalm" "true"
 Option "DisableOnPalm" "true"
 Option "Sensitivity" "0.40"
 Option "ButtonEnable" "true"
 Option "ButtonIntegrated" "true"
 Option "TapButton1" "0"
 Option "TapButton2" "0"
 Option "TapButton3" "0"
 Option "TapButton4" "0"
 Option "BottomEdge" "25"
 Option "ClickFinger1" "1"
 Option "ClickFinger2" "3"
 Option "ClickFinger3" "3"
 Option "ButtonMoveEmulate" "false"
 Option "TapDragEnable" "false"
 Option "Swipe4RightButton" "0"
 Option "Swipe4LeftButton" "0"
 Option "SwipeRightButton" "9"
 Option "SwipeLeftButton" "8"
 Option "ScrollUpButton" "4"
 Option "ScrollDownButton" "5" 
EndSection

malteo, 2016-03-24:
I can confirm this with a simpler configuration

```
Section "InputClass"
    MatchIsTouchpad "on"
    Identifier "Touchpads"
    Driver "mtrack"
    Option "Sensitivity" "0.5"
    Option "FingerHigh" "6"
    Option "FingerLow" "6"
    Option "IgnorePalm" "true"
    Option "TapButton1" "0"
    Option "TapButton2" "0"
    Option "TapButton3" "0"
    Option "TapButton4" "0"
    Option "ScrollDistance" "75"
    Option "ScrollClickTime" "10"
EndSection
```

malteo, 2016-03-25:
It seems to click when you tap and hold with two fingers

Nax, 2016-06-21:
Same proble here. All taping disabled and clicks still trigger when one finger rests on the trackpad and a 2nd one taps.

dennisfen, 2016-06-25:
I can confirm this issue on Macbook Pro Retina 13, Late 2012.
Here's my config:
```
Section "InputClass"
        MatchIsTouchpad "on"
        Identifier      "Touchpads"
        Driver          "mtrack"
        Option          "Sensitivity"           "0.50"
        Option          "IgnoreThumb"           "true"
        Option          "IgnorePalm"            "true"

        #disable tap-click
        Option          "TapButton1"            "0"
        Option          "TapButton2"            "0"
        Option          "TapButton3"            "0"
        Option          "TapButton4"            "0"

        Option          "ClickFinder1"          "1"
        Option          "ClickFinder2"          "3"
        Option          "ClickFinder3"          "2"
        Option          "BottomEdge"            "35"
        Option          "ButtonIntegrated"      "true"
        Option          "ButtonMoveEmulate"     "false"

        #natural scrolling
        Option          "ScrollDownButton"      "4"
        Option          "ScrollUpButton"        "5"
        Option          "ScrollLeftButton"      "7"
        Option          "ScrollRightButton"     "6"
EndSection
```

p2rkw, 2016-10-16:
https://github.com/p2rkw/xf86-input-mtrack/issues/15