rynbrd/xf86-input-mtrack #116: Support Apple Magic Trackpad 2

nemca, 2016-08-12:
Hi, i'm try use Apple Magic Trackpad 2 with mtrack driver, but its not works. This trackpad supported on this driver?
```
# cat /etc/X11/xorg.conf.d/10-mtrack.conf
Section "InputClass"
        Identifier      "Apple Magic Trackpad 2"
        MatchUSBID      "05ac:0265"
        Option          "Device"                "/dev/input/event3"
        Driver          "mtrack"
EndSection
```
```
[   127.483] (II) LoadModule: "mtrack"
[   127.510] (II) Loading /usr/lib64/xorg/modules/input/mtrack_drv.so
[   127.536] (II) Module mtrack: vendor="X.Org Foundation"
[   127.536]    compiled for 1.17.4, module version = 0.1.0
[   127.536]    Module class: X.Org XInput Driver
[   127.536]    ABI class: X.Org XInput driver, version 21.0
[   127.536] (II) Using input driver 'mtrack' for 'Apple Inc. Magic Trackpad 2'
[   127.536] (**) Apple Inc. Magic Trackpad 2: always reports core events
[   127.536] (**) Apple Inc. Magic Trackpad 2: always reports core events
[   127.536] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb3/3-6/3-6:1.1/0003:05AC:0265.0005/input/input6/event3"
[   127.536] (II) XINPUT: Adding extended input device "Apple Inc. Magic Trackpad 2" (type: TOUCHPAD, id 9)
[   127.536] (II) device control: init
[   127.536] (**) Option "Device" "/dev/input/event3"
[   127.536] (II) mtrack: devname: Apple Inc. Magic Trackpad 2
[   127.536] (II) mtrack: devid: 5ac 265 110
[   127.536] (II) mtrack: caps: left middle right
[   127.540] (**) Apple Inc. Magic Trackpad 2: (accel) keeping acceleration scheme 1
[   127.540] (**) Apple Inc. Magic Trackpad 2: (accel) acceleration profile 0
[   127.540] (**) Apple Inc. Magic Trackpad 2: (accel) acceleration factor: 2.000
[   127.540] (**) Apple Inc. Magic Trackpad 2: (accel) acceleration threshold: 4
[   127.540] (II) device control: on
[   127.540] (WW) Touchpad has minimal capabilities. Some features will be unavailable.
```
```
kenga ~ # egrep -A9 -B1 "Apple Inc. Magic Trackpad 2" /proc/bus/input/devices
I: Bus=0003 Vendor=05ac Product=0265 Version=0110
N: Name="Apple Inc. Magic Trackpad 2"
P: Phys=usb-0000:00:14.0-6/input1
S: Sysfs=/devices/pci0000:00/0000:00:14.0/usb3/3-6/3-6:1.1/0003:05AC:0265.0005/input/input6
U: Uniq=CC260260HFZG61CAY
H: Handlers=mouse0 event3
B: PROP=0
B: EV=17
B: KEY=70000 0 0 0 0
B: REL=3
B: MSC=10
```
```
# uname -r
4.4.6-gentoo
```

BlueDragonX, 2016-08-12:
It's not been treated with this device.

nemca, 2016-08-12:
It's a pity.

closed 2016-08-12 by nemca

BlueDragonX, 2016-08-13:
*tested

nemca, 2016-08-13:
Can I help in the testing of this trackpad? I'd like to make it work under Linux.