rynbrd/xf86-input-mtrack #117: Xorg not using Mtrack

jrussell88, 2016-12-30:
I installed the xserver-xorg-input-mtrack_0.4.1_amd64.deb package on my Ubuntu 16.10 x64 Macbook Retina Pro (MacBookPro11,2) and added a section to /usr/share/X11/xorg.conf.d/50-synaptics.conf.

This didn't change the behaviour of the touchpad. 

As Xorg is running in rootless mode I added myself to the Input group. 

/var/logXorg.0.log refers to InputClass evdev rather than Mtrack:
`[    12.080] (II) config/udev: Adding input device bcm5974 (/dev/input/event11)
[    12.080] (**) bcm5974: Applying InputClass "evdev touchpad catchall"
[    12.080] (**) bcm5974: Applying InputClass "touchpad catchall"
[    12.080] (**) bcm5974: Applying InputClass "Default clickpad buttons"
[    12.080] (**) bcm5974: Applying InputClass "Disable clickpad buttons on Apple touchpads"
[    12.080] (**) bcm5974: Applying InputClass "Touchpads"
[    12.080] (**) bcm5974: Applying InputClass "libinput touchpad catchall"
[    12.080] (II) Using input driver 'libinput' for 'bcm5974'
[    12.080] (**) bcm5974: always reports core events
[    12.080] (**) Option "Device" "/dev/input/event11"
[    12.080] (**) Option "_source" "server/udev"
[    12.081] (II) input device 'bcm5974', /dev/input/event11 is tagged by udev as: Touchpad
[    12.081] (II) input device 'bcm5974', /dev/input/event11 is a touchpad
[    12.128] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-12/1-12:1.2/input/input14/event11"
[    12.128] (II) XINPUT: Adding extended input device "bcm5974" (type: TOUCHPAD, id 14)
[    12.128] (**) Option "AccelerationScheme" "none"
[    12.128] (**) bcm5974: (accel) selected scheme none/0
[    12.128] (**) bcm5974: (accel) acceleration factor: 2.000
[    12.128] (**) bcm5974: (accel) acceleration threshold: 4
[    12.129] (II) input device 'bcm5974', /dev/input/event11 is tagged by udev as: Touchpad
[    12.129] (II) input device 'bcm5974', /dev/input/event11 is a touchpad
[    12.129] (II) config/udev: Adding input device bcm5974 (/dev/input/mouse2)
[    12.129] (**) bcm5974: Ignoring device from InputClass "touchpad ignore duplicates"
`

/usr/share/X11/xorg.conf.d/11-evdev-trackpoint.conf reads:
`# trackpoint users want wheel emulation

Section "InputClass"
	Identifier	"trackpoint catchall"
	MatchIsPointer	"true"
	MatchProduct	"TrackPoint|DualPoint Stick"
	MatchDevicePath	"/dev/input/event*"
	Option	"Emulate3Buttons"	"true"
	Option	"EmulateWheel"	"true"
	Option	"EmulateWheelButton"	"2"
	Option	"XAxisMapping"	"6 7"
	Option	"YAxisMapping"	"4 5"
EndSection`

What should I do to ensure that Mtrack settings are used rather than the generic evdev settings?

jrussell88, 2016-12-31:
10-evdev.conf contains the section which Xorg appears to be using:

> Section "InputClass"
>         Identifier "evdev touchpad catchall"
>         MatchIsTouchpad "on"
>         MatchDevicePath "/dev/input/event*"
>         Driver "evdev"
> EndSection

As Xorg processes config files in ascending alphabetical order, I moved the Mtrack section from 50-synaptics.conf to a 55-mtrack.conf file: 

> # Match Apple bcm5974 touchpad
> 
> Section "InputClass"
> 	MatchIsTouchpad "on"
> 	Identifier "Apple touchpad"
>	MatchVendor "05ac"
> 	MatchProduct "Apple|bcm5974"
> 	Driver "mtrack"
> 
> # 30-12-2016 Mtrack settings from http://yarenty.blogspot.com/2014/08/how-to-fix-macbook-pro-touchpad-on.html
> 	Option "Sensitivity" "0.65"
> 	Option "FingerHigh" "12"
> 	Option "FingerLow" "4"
> 	Option "IgnoreThumb" "true"
> 	Option "IgnorePalm" "true"
> 	Option "TapButton1" "0"
> 	Option "TapButton2" "0"
> 	Option "TapButton3" "0"
> 	Option "TapButton4" "0"
> 	Option "ClickFinger1" "3"
> 	Option "ClickFinger2" "2"
> 	Option "ClickFinger3" "3"
> 	Option "ButtonMoveEmulate" "true"
> 	Option "ButtonIntegrated" "true"
> 	Option "ClickTime" "25"
> 	Option "BottomEdge" "40"
> 	Option "SwipeLeftButton" "8"
> 	Option "SwipeRightButton" "9"
> 	Option "SwipeUpButton" "0"
> 	Option "SwipeDownButton" "0"
> 	Option "ScrollDistance" "75"
> 	Option "ScrollUpButton" "4"
> 	Option "ScrollDownButton" "5"
> 	Option "ThumbSize" "35"
> 	Option "PalmSize" "55"
> 	Option "DisableOnThumb" "false"
> 	Option "DisableOnPalm" "true"
> 	Option "SwipeDistance" "1000"
> 	Option "ScrollLeftButton" "7"
> 	Option "ScrollRightButton" "6"
> 	Option "AccelerationProfile" "2"
> 	Option "ConstantDeceleration" "2.0" # Decelerate endspeed
> 	Option "AdaptiveDeceleration" "2.0" # Decelerate slow movements
> 
> EndSection

I took the vendor id from lsusb:

> Bus 001 Device 006: ID 05ac:0263 Apple, Inc. Apple Internal Keyboard / Trackpad (MacBook Retina)

However Xorg.0.log still refers to the libinput driver for the touchpad rather than Mtrack.

piggyslasher, 2017-09-12:
I had an error in my mtrack.conf and it was falling back to libinput.

Try grepping the log with `| grep ouchpad`.