rynbrd/xf86-input-mtrack #118: Driver would not load on MBP Early 2015

slashrsm, 2017-01-06:
First things first :) I did check existing issues with the same error message and made sure I am in the `input` group:

````
➜  ~  groups
sys adm lp wheel input usbmux docker git slashrsm
````

My config is:

````
Section "InputClass"
    MatchIsTouchpad "on"
    Identifier "Touchpads"
    Driver "mtrack"
EndSection
````

And the most relevant part of the xorg log is:
````
[  1029.046] (II) Module mtrack: vendor="X.Org Foundation"
[  1029.046] 	compiled for 1.18.4, module version = 0.2.0
[  1029.046] 	Module class: X.Org XInput Driver
[  1029.046] 	ABI class: X.Org XInput driver, version 22.1
[  1029.046] (II) Using input driver 'mtrack' for 'bcm5974'
[  1029.046] (**) bcm5974: always reports core events
[  1029.046] (**) bcm5974: always reports core events
[  1029.046] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.2/input/input8/mouse1"
[  1029.046] (II) XINPUT: Adding extended input device "bcm5974" (type: TOUCHPAD, id 14)
[  1029.046] (II) device control: init
[  1029.046] (**) Option "Device" "/dev/input/mouse1"
[  1029.046] (EE) mtrack: cannot configure device
[  1029.046] (EE) Couldn't init device "bcm5974"
[  1029.046] (II) UnloadModule: "mtrack"
````

I would be very thankful for any clues. 

Full log:

````
[  1028.976] (--) Log file renamed from "/home/slashrsm/.local/share/xorg/Xorg.pid-7383.log" to "/home/slashrsm/.local/share/xorg/Xorg.0.log"
[  1028.976] 
X.Org X Server 1.18.4
Release Date: 2016-07-19
[  1028.976] X Protocol Version 11, Revision 0
[  1028.976] Build Operating System: Linux 4.5.4-1-ARCH x86_64 
[  1028.976] Current Operating System: Linux pineapple 4.8.13-1-ARCH #1 SMP PREEMPT Fri Dec 9 07:24:34 CET 2016 x86_64
[  1028.976] Kernel command line: BOOT_IMAGE=/vmlinuz-linux cryptdevice=/dev/sda3:Pineapple root=/dev/Pineapple/root rw quiet libata.atapi_passthru16=0
[  1028.976] Build Date: 19 July 2016  05:54:24PM
[  1028.976]  
[  1028.976] Current version of pixman: 0.34.0
[  1028.976] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[  1028.976] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[  1028.976] (==) Log file: "/home/slashrsm/.local/share/xorg/Xorg.0.log", Time: Fri Jan  6 22:16:45 2017
[  1028.976] (==) Using config directory: "/etc/X11/xorg.conf.d"
[  1028.976] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[  1028.976] (==) No Layout section.  Using the first Screen section.
[  1028.976] (==) No screen section available. Using defaults.
[  1028.976] (**) |-->Screen "Default Screen Section" (0)
[  1028.976] (**) |   |-->Monitor "<default monitor>"
[  1028.977] (==) No monitor specified for screen "Default Screen Section".
	Using a default monitor configuration.
[  1028.977] (==) Automatically adding devices
[  1028.977] (==) Automatically enabling devices
[  1028.977] (==) Automatically adding GPU devices
[  1028.977] (==) Max clients allowed: 256, resource mask: 0x1fffff
[  1028.977] (WW) `fonts.dir' not found (or not valid) in "/usr/share/fonts/100dpi/".
[... 291 lines cut ...]
[  1029.046] (**) bcm5974: always reports core events
[  1029.046] (**) bcm5974: always reports core events
[  1029.046] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.2/input/input8/mouse1"
[  1029.046] (II) XINPUT: Adding extended input device "bcm5974" (type: TOUCHPAD, id 14)
[  1029.046] (II) device control: init
[  1029.046] (**) Option "Device" "/dev/input/mouse1"
[  1029.046] (EE) mtrack: cannot configure device
[  1029.046] (EE) Couldn't init device "bcm5974"
[  1029.046] (II) UnloadModule: "mtrack"
[  1029.046] (II) config/udev: Adding input device HDA Intel PCH Headphone (/dev/input/event10)
[  1029.046] (II) No input driver specified, ignoring this device.
[  1029.046] (II) This device may have been added with another device file.
[  1029.046] (II) config/udev: Adding input device HDA Intel PCH SPDIF (/dev/input/event11)
[  1029.046] (II) No input driver specified, ignoring this device.
[  1029.046] (II) This device may have been added with another device file.
[  1029.047] (II) config/udev: Adding input device HDA Intel PCH Mic (/dev/input/event9)
[  1029.047] (II) No input driver specified, ignoring this device.
[  1029.047] (II) This device may have been added with another device file.
[  1029.047] (II) config/udev: Adding input device PC Speaker (/dev/input/event7)
[  1029.047] (II) No input driver specified, ignoring this device.
[  1029.047] (II) This device may have been added with another device file.
[  1029.235] (II) intel(0): EDID vendor "APP", prod id 41002
[  1029.235] (II) intel(0): Printing DDC gathered Modelines:
[  1029.235] (II) intel(0): Modeline "2560x1600"x0.0  268.50  2560 2608 2640 2720  1600 1603 1609 1646 +hsync -vsync (98.7 kHz eP)
````

duncanfinney, 2017-03-08:
I'm seeing the same thing. I am running Manjaro linux on a Macbook Pro 13 inch Early 2015.

MacbookPro 12,1

svenper, 2017-06-29:
Same issue for me with MacBookPro`12,1` on Void Linux

Jens0512, 2018-09-22:
Same issue here.On MacBookAir4,1 1.0
```sh
% pacman -Qi xf86-input-mtrack-git
Name            : xf86-input-mtrack-git
Version         : 0.5.0.r5.gf09dc9b-1
Description     : A multitouch X driver using the kernel MT protocol
Architecture    : x86_64
URL             : http://github.com/p2rkw/xf86-input-mtrack
Licenses        : GPL
Groups          : None
Provides        : xf86-input-mtrack
Depends On      : mtdev  libxss
Optional Deps   : None
Required By     : None
Optional For    : None
Conflicts With  : xf86-input-mtrack  xf86-input-synaptics
Replaces        : None
Installed Size  : 90.00 KiB
Packager        : Unknown Packager
Build Date      : Thu 02 Aug 2018 18:05:30 CEST
Install Date    : Thu 02 Aug 2018 18:06:49 CEST
Install Reason  : Explicitly installed
Install Script  : Yes
Validated By    : None

% uname -mors
Linux 4.18.7-arch1-1-ARCH x86_64 GNU/Linux
```

svenper, 2018-09-22:
IIRC a workaround for this is to run these commands at boot:

```
modprobe -r usbmouse
modprobe -r bcm5974
modprobe bcm5974
```