rynbrd/xf86-input-mtrack #123: fatal error: mtdev-mapping.h / CentOS

GIT-STATA, 2017-10-23:
Hi,

thanks a lot for all that great work! I would love to use the driver as well! 
I am trying to use the xf86-input-mtrack on CentOS7.

I do get a  **fatal error: mtdev-mapping.h: No such file or directory**.
***So what can I do to install the driver?*** 

I do really appreciate all kind of help! 
**Thanks a lot!!!**

## Error 
make  all-am           
make[1]: Entering directory `/tmp/xf86-input-mtrack'         

  CC       src/capabilities.lo             
In file included from ./include/capabilities.h:25:0,                        
                 from src/capabilities.c:22:                                 

./include/common.h:35:27: fatal error: mtdev-mapping.h: No such file or directory              
 #include <mtdev-mapping.h>
                           ^
compilation terminated.          

make[1]: *** [src/capabilities.lo] Fehler 1           
make[1]: Leaving directory `/tmp/xf86-input-mtrack'              
make: *** [all] Fehler 2               

## What I did on CentOS7 

sudo yum install gcc xorg-x11-util-macros xorg-x11-server-devel libXext-devel libXi-devel libXrandr-devel libXinerama-devel libudev-devel

cd /tmp
git clone https://github.com/p2rkw/xf86-input-mtrack.git
cd xf86-input-mtrack
./configure --with-xorg-module-dir=/usr/lib64/xorg/modules
make && make install

p2rkw, 2017-10-28:
sudo yum install mtdev-devel

GIT-STATA, 2017-11-16:
## I still get errors

### I do start with... 
./configure --with-xorg-module-dir=/usr/lib64/xorg/modules       

make && make install

### 

> 

> `make  all-am
make[1]: Entering directory `/tmp/xf86-input-mtrack'
  CCLD     mtrack-test
src/mtrack_test-hwstate.o: In function `hwstate_modify':
/tmp/xf86-input-mtrack/src/hwstate.c:117: undefined reference to `mtdev_get'
src/mtrack_test-mtouch.o: In function `mtouch_open':
/tmp/xf86-input-mtrack/src/mtouch.c:42: undefined reference to `mtdev_open'
src/mtrack_test-mtouch.o: In function `mtouch_close':
/tmp/xf86-input-mtrack/src/mtouch.c:70: undefined reference to `mtdev_close'
collect2: error: ld returned 1 exit status
make[1]: *** [mtrack-test] Fehler 1
make[1]: Leaving directory `/tmp/xf86-input-mtrack'
make: *** [all] Fehler 2
`

p2rkw, 2017-11-16:
sudo yum install mtdev

hdenman, 2018-02-11:
GIT-STATA - I suggest running 'make clean' and rerunning './configure' after install any new system package via yum or apt; this fixed your second error for me.

Many thanks to p2rkw et al for mtrack!

dulm, 2018-09-26:
i'v got same error with GIT-STATA  after install mtdev-devel .

make  all-am
make[1]: Entering directory '/home/lqk/Downloads/xf86-input-mtrack'
  CCLD     mtrack-test
src/mtrack_test-hwstate.o: In function `hwstate_modify':
/home/lqk/Downloads/xf86-input-mtrack/src/hwstate.c:126: undefined reference to `mtdev_get'
src/mtrack_test-mtouch.o: In function `mtouch_open':
/home/lqk/Downloads/xf86-input-mtrack/src/mtouch.c:42: undefined reference to `mtdev_open'
src/mtrack_test-mtouch.o: In function `mtouch_close':
/home/lqk/Downloads/xf86-input-mtrack/src/mtouch.c:70: undefined reference to `mtdev_close'
collect2: error: ld returned 1 exit status
Makefile:565: recipe for target 'mtrack-test' failed
make[1]: *** [mtrack-test] Error 1
make[1]: Leaving directory '/home/lqk/Downloads/xf86-input-mtrack'
Makefile:419: recipe for target 'all' failed
make: *** [all] Error 2

voneAbides, 2020-02-23:
Same error in Fedora 31 with KDE Plasma

lucasmsoares96, 2020-05-26:
on Xubuntu I solved it like this:
`sudo apt search mtdev`
` sudo apt install for-all-search-results`
`sudo make clean`
`sudo ./configure`