rynbrd/xf86-input-mtrack #125: Weird scrolling issue

wpovell, 2017-12-13:
I've found that if I scroll, lift up, and scroll again, I start scrolling super fast until I lift up again. The normal scroll speed is fine. Any idea what might cause this? I haven't seen any relavant settings in the documentation that would control this.

slang800, 2018-10-03:
I'm having the same issue. I didn't notice this until I enabled scroll coasting (which makes the issue way worse), but I can reproduce it without scroll coasting enabled. This is my config:

```
Section "InputClass"
  Identifier "touchpad"
  Driver "mtrack"
  MatchIsTouchpad "on"
  MatchDevicePath "/dev/input/event*"
  Option "Sensitivity" "0.3"
  Option "TapButton1" "1"
  Option "TapButton2" "0"
  Option "TapButton3" "0"
  Option "TapButton4" "0"
  Option "ClickFinger1" "1"
  Option "ClickFinger2" "3"
  Option "ClickFinger3" "0"
  Option "ButtonMoveEmulate" "false"
  Option "FingerHigh" "10"
  Option "FingerLow" "1"
  Option "IgnoreThumb" "true"
  Option "IgnorePalm" "true"
  Option "TapDragEnable" "false"
  Option "ScrollDownButton" "4"
  Option "ScrollUpButton" "5"
  Option "ScrollLeftButton" "7"
  Option "ScrollRightButton" "6"
  Option "ScrollCoastDuration" "600"
  Option "ScrollCoastEnableSpeed" "0.05"
EndSection
```

artefom, 2021-03-10:
Can indeed confirm that this is the case. Looks like a bug. Possibly coasting timer does not get disabled after tapping again during coasting, so scrolling gets applied twice?