rynbrd/xf86-input-mtrack #129: Cannot load mtrack

dcrystalj, 2018-10-28:
After successful compilation and  installation mtrack && reboot, I cannot load mtrack drivers. I am running latest deepin linux, debian based distro on lenovo yoga 2 pro
```
ls /usr/share/X11/xorg.conf.d
➜  xf86-input-mtrack git:(master) ls /usr/share/X11/xorg.conf.d/                
10-amdgpu.conf  10-quirks.conf  10-radeon.conf  30-mtrack.conf  35-mtrack.conf  40-libinput.conf  50-mtrack.conf  70-wacom.conf
```

comman `egrep -i "mtrack" /var/log/Xorg.0.log ` outputs nothing

whole log:
```
[     7.105] 
X.Org X Server 1.20.0
X Protocol Version 11, Revision 0
[     7.106] Build Operating System: Linux 4.9.0-6-amd64 x86_64 Debian
[     7.106] Current Operating System: Linux dcrystalj-PC 4.15.0-29deepin-generic #31 SMP Fri Jul 27 07:12:08 UTC 2018 x86_64
[     7.106] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-4.15.0-29deepin-generic root=UUID=113ecab9-2fe0-4d9e-8a1d-e964b312fe8b ro splash quiet
[     7.106] Build Date: 24 May 2018  04:23:27PM
[     7.106] xorg-server 2:1.20.0-2 (https://www.debian.org/support) 
[     7.106] Current version of pixman: 0.34.0
[     7.106] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[     7.106] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[     7.106] (==) Log file: "/var/log/Xorg.0.log", Time: Sun Oct 28 09:12:44 2018
[     7.174] (==) Using config directory: "/etc/X11/xorg.conf.d"
[     7.174] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[     7.241] (==) No Layout section.  Using the first Screen section.
[     7.241] (==) No screen section available. Using defaults.
[     7.241] (**) |-->Screen "Default Screen Section" (0)
[     7.241] (**) |   |-->Monitor "<default monitor>"
[     7.241] (==) No monitor specified for screen "Default Screen Section".
	Using a default monitor configuration.
[     7.241] (==) Automatically adding devices
[     7.241] (==) Automatically enabling devices
[     7.241] (==) Automatically adding GPU devices
[     7.241] (==) Max clients allowed: 256, resource mask: 0x1fffff
[     7.241] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[     7.241] 	Entry deleted from font path.
[     7.241] (==) FontPath set to:
	/usr/share/fonts/X11/misc,
	/usr/share/fonts/X11/100dpi/:unscaled,
	/usr/share/fonts/X11/75dpi/:unscaled,
	/usr/share/fonts/X11/Type1,
	/usr/share/fonts/X11/100dpi,
	/usr/share/fonts/X11/75dpi,
	built-ins
[     7.241] (==) ModulePath set to "/usr/lib/xorg/modules"
[     7.241] (II) The server relies on udev to provide the list of input devices.
	If no devices become available, reconfigure udev or disable AutoAddDevices.
[     7.241] (II) Loader magic: 0x562848400de0
[     7.241] (II) Module ABI versions:
[     7.241] 	X.Org ANSI C Emulation: 0.4
[     7.241] 	X.Org Video Driver: 24.0
[     7.241] 	X.Org XInput driver : 24.1
[     7.241] 	X.Org Server Extension : 10.0
[     7.242] (++) using VT number 1

[     7.242] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[     7.242] (II) xfree86: Adding drm device (/dev/dri/card0)
[     7.244] (--) PCI:*(0@0:2:0) 8086:0a16:17aa:3978 rev 9, Mem @ 0xb0000000/4194304, 0xa0000000/268435456, I/O @ 0x00003000/64, BIOS @ 0x????????/131072
[     7.244] (II) LoadModule: "glx"
[     7.244] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[     7.244] (II) Module glx: vendor="X.Org Foundation"
[     7.244] 	compiled for 1.20.0, module version = 1.0.0
[     7.244] 	ABI class: X.Org Server Extension, version 10.0
[     7.244] (==) Matched modesetting as autoconfigured driver 0
[     7.244] (==) Matched fbdev as autoconfigured driver 1
[     7.244] (==) Matched vesa as autoconfigured driver 2
[... 437 lines cut ...]
[     7.692] (II) UnloadModule: "libinput"
[     7.692] (II) config/udev: Adding input device PC Speaker (/dev/input/event7)
[     7.692] (II) No input driver specified, ignoring this device.
[     7.692] (II) This device may have been added with another device file.
[     7.856] (II) modeset(0): EDID vendor "SDC", prod id 16970
[     7.856] (II) modeset(0): Printing DDC gathered Modelines:
[     7.856] (II) modeset(0): Modeline "3200x1800"x0.0  361.31  3200 3248 3280 3316  1800 1802 1807 1816 -hsync -vsync (109.0 kHz eP)
[     7.856] (II) modeset(0): Modeline "3200x1800"x0.0  361.31  3200 3248 3280 3680  1800 1802 1807 2045 -hsync -vsync (98.2 kHz e)
[     7.874] (II) modeset(0): EDID vendor "SDC", prod id 16970
[     7.874] (II) modeset(0): Printing DDC gathered Modelines:
[     7.874] (II) modeset(0): Modeline "3200x1800"x0.0  361.31  3200 3248 3280 3316  1800 1802 1807 1816 -hsync -vsync (109.0 kHz eP)
[     7.874] (II) modeset(0): Modeline "3200x1800"x0.0  361.31  3200 3248 3280 3680  1800 1802 1807 2045 -hsync -vsync (98.2 kHz e)
[    13.283] (II) modeset(0): EDID vendor "SDC", prod id 16970
[    13.283] (II) modeset(0): Printing DDC gathered Modelines:
[    13.283] (II) modeset(0): Modeline "3200x1800"x0.0  361.31  3200 3248 3280 3316  1800 1802 1807 1816 -hsync -vsync (109.0 kHz eP)
[    13.284] (II) modeset(0): Modeline "3200x1800"x0.0  361.31  3200 3248 3280 3680  1800 1802 1807 2045 -hsync -vsync (98.2 kHz e)
[    13.300] (II) modeset(0): EDID vendor "SDC", prod id 16970
[    13.300] (II) modeset(0): Printing DDC gathered Modelines:
[    13.300] (II) modeset(0): Modeline "3200x1800"x0.0  361.31  3200 3248 3280 3316  1800 1802 1807 1816 -hsync -vsync (109.0 kHz eP)
[    13.300] (II) modeset(0): Modeline "3200x1800"x0.0  361.31  3200 3248 3280 3680  1800 1802 1807 2045 -hsync -vsync (98.2 kHz e)
[    14.856] (II) modeset(0): EDID vendor "SDC", prod id 16970
[    14.856] (II) modeset(0): Printing DDC gathered Modelines:
[    14.856] (II) modeset(0): Modeline "3200x1800"x0.0  361.31  3200 3248 3280 3316  1800 1802 1807 1816 -hsync -vsync (109.0 kHz eP)
[    14.856] (II) modeset(0): Modeline "3200x1800"x0.0  361.31  3200 3248 3280 3680  1800 1802 1807 2045 -hsync -vsync (98.2 kHz e)
```

alexkohler, 2018-12-15:
I am also running into a similar issue.

54D-R0807, 2018-12-15:
Me too, running openSUSE Leap 15 (tested on: 4.12.14-lp150.12.22, 4.12.14-lp150.12.25 and 4.12.14-lp150.12.28). I started to notice this a week ago. Not sure though, if the problem existed before.

`modprobe mtrack
modprobe: FATAL: Module mtrack not found in directory /lib/modules/4.12.14-lp150.12.28-default`

`cat /var/log/Xorg.0.log | grep mtrack
[    21.721] (II) LoadModule: "mtrack"
[    21.721] (WW) Warning, couldn't open module mtrack
[    21.721] (EE) Failed to load module "mtrack" (module does not exist, 0)
[    21.721] (EE) No input driver matching 'mtrack' `

I have already tried reinstalling the software without any differences.

jcrben, 2019-01-26:
You might consider trying this fork: https://github.com/p2rkw/xf86-input-mtrack