rynbrd/xf86-input-mtrack #135: Touchpad disabled at login screen

retesta, 2020-05-08:
After successfully installing and configuring mtrack on my LG Gram 2018, I am unable to use the touchpad on the login screen. It won't register clicks nor movement. Plugging in a mouse works on the login screen.

After logging in everything works properly as it should. The issue only happens to the touchpad at the login screen.

Running Ubuntu 18.04 LTS

closed 2020-05-08 by retesta

gerrywastaken, 2020-05-10:
Why did you open and then close the issue on the same day without saying anything? @retesta

retesta, 2020-06-14:
Sorry, I realized I was using this version after I opened the issue: https://github.com/p2rkw/xf86-input-mtrack

I should I have said something when I closed it though. I apologize

gerrywastaken, 2020-06-15:
Thanks for the update retesta, I appreciate it. :D