rynbrd/xf86-input-mtrack #136: Four finger gestures do not work on a Dell XPS 13 9300

hisbaan, 2020-06-11:
Hi,

I have a peculiar issue and I wasn't able to find anything in the other issues that fixed my issue. I can use 1, 2, and 3 finger gestures on my trackpad but 4 finger ones just don't seem to work. I was using the exact same configuration on my old laptop (MacBook Air 2015) and it was working just fine. I'm running arch linux and installed mtrack from the AUR (aur: xf86-input-mtrack). The multitouch gestures work just fine in windows so I'm not really sure what's wrong. 

Here's the output of `pacman -Qi xf86-input-mtrack`
```
Name            : xf86-input-mtrack
Version         : 0.5.0-3
Description     : An Xorg driver for multitouch trackpads and clickpads
Architecture    : x86_64
URL             : https://github.com/p2rkw/xf86-input-mtrack
Licenses        : GPL2
Groups          : None
Provides        : None
Depends On      : mtdev  libxss
Optional Deps   : None
Required By     : None
Optional For    : None
Conflicts With  : None
Replaces        : None
Installed Size  : 61.91 KiB
Packager        : Unknown Packager
Build Date      : Mon 01 Jun 2020 08:15:08 PM EDT
Install Date    : Wed 10 Jun 2020 09:32:44 PM EDT
Install Reason  : Explicitly installed
Install Script  : No
Validated By    : None
```

And here's my configuration file
```
Section "InputClass"
        MatchIsTouchpad "on"
        Identifier      "Touchpads"
        MatchDevicePath "/dev/input/event*"
        Driver          "mtrack"
        # The faster you move, the more distance pointer will travel, using "polynomial" profile
        Option          "AccelerationProfile" "2"
        # Tweak cursor movement speed with this
        Option          "Sensitivity" "0.9"
        # Pressure at which a finger is detected as a touch
        Option          "FingerHigh" "5"
        # Pressure at which a finger is detected as a release
        Option          "FingerLow" "5"
        # I often use thumb to press down the physical button, so let's not ignore it
        Option          "IgnoreThumb" "false"
        Option          "ThumbRatio" "70"
        Option          "ThumbSize" "25"
        # Ignore palm, with palm takes up to 30% of your touch pad
        Option          "IgnorePalm" "true"
        Option          "PalmSize" "15"
        # Trigger mouse button when tap: 1 finger - left click, 2 finger - right click, 3 - middle click
        Option          "TapButton1" "1"
        Option          "TapButton2" "3"
        Option          "TapButton3" "2"
        Option          "TapButton4" "0"
        Option          "ClickTime" "25"
        # Disable tap-to-drag, we're using three finger drag instead
        Option          "TapDragEnable" "false"
        # While touching the touch pad with # fingers, press the touchpad physical click button
        Option          "ClickFinger1" "1"
        Option          "ClickFinger2" "3"
        Option          "ClickFinger3" "2"
        Option          "ButtonMoveEmulate" "false"
        Option          "ButtonIntegrated" "true"
        # The momentum after scroll fingers released
        Option          "ScrollCoastDuration" "300"
        Option          "ScrollCoastEnableSpeed" ".1"
        # Natural scrolling with two fingers
        Option          "ScrollSmooth" "true"
        Option          "ScrollUpButton" "4"
        Option          "ScrollDownButton" "5"
        Option          "ScrollLeftButton" "6"
        Option          "ScrollRightButton" "7"
        # Tweak scroll sensitivity with ScrollDistance, don't touch ScrollSensitivity
        Option          "ScrollDistance" "25"
        Option          "ScrollClickTime" "10"
        # Three finger drag
        Option          "SwipeDistance" "1"
        Option          "SwipeLeftButton" "1"
        Option          "SwipeRightButton" "1"
        Option          "SwipeUpButton" "1"
        Option          "SwipeDownButton" "1"
        Option          "SwipeClickTime" "0"
        Option          "SwipeSensitivity" "1000"
        # Four finger swipe, 8 & 9 are for browsers navigating back and forth respectively
        Option          "Swipe4LeftButton" "8"
        Option          "Swipe4RightButton" "9"
        # Mouse button >= 10 are not used by Xorg, so we'll map them with xbindkeys and xdotool later
        Option          "Swipe4UpButton" "11"
        Option          "Swipe4DownButton" "10"
        # Mouse buttons triggered by 2-finger pinching gesture
        Option          "ScaleDistance" "300"
        Option          "ScaleUpButton" "12"
        Option          "ScaleDownButton" "13"
        # Mouse buttons trigger by 2-finger rotating gesture, disabled to enhance the pinch gesture
        Option          "RotateLeftButton" "0"
        Option          "RotateRightButton" "0"
EndSection
```

Thanks for the help :)

misketmuskat, 2020-10-16:
Do you have any action assigned to four finger swipe (Button10 & Button11)?
I am using Openbox in my Macbook Air, and I am able to assign action such as "ToggleShowDesktop" to these gestures from Openbox's "rc.xml".
I guess you can assign actions to these keys by using xdotool.

hisbaan, 2020-10-16:
No, I don't have any actions set. I use bspwm with sxhkd. I've tried using xev to see if the four finger swipe was being detected but no luck there.